bdring/4_Axis_SPI_CNC #2: where is cpu_map.h ?

cxandy, 2019-10-19:
thanks.

bdring, 2019-10-19:
This board used to require a fork of Grbl_ESP32 and a fork of TMCStepper. 

The original TMCStepper has been updated to support daisy chained SPI and Grbl_ESP32 now uses that version. The old fork of Grbl_ESP32 was removed.

Support for daisy chained SPI can now be added to the master branch, but it has not been done yet. It is on my todo list, but behind a few other priorities. I hope to have it done by the end of the month.

shreeramlive, 2019-11-04:
In the schematic I noticed that the **Step/Dir** pins are physically connected to the ESP32 board. Do we need them for **SPI connection**, if they are not needed we can add some switches like Hold, Resume.
And the connection to the limit switches are without any **filter capacitors** unlike the [Grbl_ESP32](https://github.com/bdring/Grbl_ESP32_Development_Controller/blob/master/source/V3p5/schematic_esp32_cnc_test_v3p5.pdf) development board, are they necessary for the ESP32 boards?

bdring, 2019-11-04:
**Step/Dir** They are required. SPI primarily sets the features of the driver. It does have the ability to run motion profiles, but not step and direction. You could eliminate the driver **enable** pin. All the drivers share this one pin, but support would need to be added to Grbl for it.

**R/C filters on inputs** It is not typically required for smaller machines. Grbl also has debouncing that can be used. If you are using a larger machine with a high current spindle or laser, you might want to use them.

shreeramlive, 2019-11-09:
Hi, I have created a schematic with a slightly different pin map. **RC filters** are added for limit switches to be able to use with large machines.

[image]

[Schematic_esp32 grblcnc bk_Sheet_1 edit copy_20191109121537.pdf](https://github.com/bdring/4_Axis_SPI_CNC/files/3826991/Schematic_esp32.grblcnc.bk_Sheet_1.edit.copy_20191109121537.pdf)

A **TXS0104 level shifter** is also added to provide four 5V output pins to drive servo axes and spindle.
When spindle enable and coolant is not needed, can we use the `level shifter` pins as **inputs** for other functions like hold, resume?

I'm trying to create a prototype with this design, Do I have change any pin configuration?
Thanks

bdring, 2019-11-09:
As drawn, those are not R/C filters. There is no "R".

I suggest grounding the CLK pin on the drivers to make it compatible with TMC5160