bdring/4_Axis_SPI_CNC #3: Connection issue with daisy chained TMC5160's

shreeramlive, 2019-12-03:
Hi I tried to connect 4 **TMC5160** in daisy chain. However I could not get the motor moving.  There seems to be a **SPI problem** with the TMC5160s. I have also _grounded_ the CLK pin

However I'm pretty sure that the boards works because when I connected the` A4988 drivers `it was working perfectly. 
[image]
Here's the ESP32 connection diagram. 

Is there any way to debug the TMC connection like in marlin (**M122**)?
I think there is a problem with the TMC5160 drivers, a fix for SKR1.3 was provided [here](https://github.com/teemuatlut/TMCStepper/pull/63/commits/c481cccf542f3243c2c0594f5cf765fde2dd22d0)
This was a patch for the SKR1.3 (LPC1768), Is there anything similar to get it working with the ESP32.
Thank you

BTW: This board is designed to support both standalone drivers and SPI drivers. That's how I was able to test the A4988 drivers

shreeramlive, 2019-12-03:
[image]

[image]

Here are some photos, Everything works well except the TMC5160 SPI connection.
P.S: Board named after the project creator @bdring.

bdring, 2019-12-03:
Can you provide the complete schematic, or at least the TMC5160 connections?

shreeramlive, 2019-12-03:
Yeah sure,
Most of the connections are similar to your 4 axis SPI board.

[image]

The extra headers are allow us to configure SD and SPI connections
Since I'm using TMC5160, I have grounded the CLK to ground

[sch2.0.pdf](https://github.com/bdring/4_Axis_SPI_CNC/files/3917623/sch2.0.pdf)

bdring, 2019-12-03:
What is the purpose of the stepper headers connected to 3.3V and CLK?

[image]

Try connecting CLK to ground.

shreeramlive, 2019-12-17:
**Thanks @bdring**, I have tried connecting the **CLK to ground**. Nevermind the headers in the schematic. Those extra headers are used to configure **Microsteps or SPI** connections with the help of **jumpers.**

Still no success with the TMC5160 stepsticks, I'm kind of confident that the _issue is with the TMC5160 drivers_. I have ordered some `TMC2130's` for testing. I will post updates when I receive them.

Is there any way to **debug** the `SPI connection` with the drivers?

Baramykin, 2020-12-24:
Hi,
I don't understand how to connect TMC5160 8 pcs to Teensy 4.1 in UART mode.
I can't find the schematics.
I want to use TMC5160 V1.2 drivers.
Help me if possible, please.

creatormir, 2021-05-11:
I've come to see the serial connection.
And about the SPI, I think the problem is the StepStick board. The SPI is blocked. We need to cut the track. That's where the https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/issues/8 discussed it.

The place to cut. It helped me.
[image]