bdring/4_Axis_SPI_CNC #4: where to buy

haklein, 2020-02-25:
Hi @bdring,

would it be possible to buy one of the testing boards? I've some experience with TMCs from my Powder'o'matic project (https://github.com/haklein/Powder-O-Matic). I'd like to help with testing and convert a CNC controller (from an RS-CNC32) with external drivers to TMC5160.  

br Hari

bdring, 2020-02-25:
I have a new version 1.3 that I plan to sell on Tindie soon. It will probably be at least 4-5 weeks before they are ready due to the problems in China.

I have 3 hand built prototypes that I would be happy to sell for $60 each. They would not include drivers or ESP32.

[image]

haklein, 2020-02-25:
I'd happily take one of these prototypes (I'd assume all features are tested and it's the same as the final version?).

bdring, 2020-02-25:
The design is 100% tested. Also, some new features have been added to Grbl_ESP32 to allow any these axes to be combined (2 motor on one axis).

Please sent $60 + $5 US shipping or $14 International to the PayPal donation link on the first page of this repo. Be sure to add a note about reason for the donation and include the shipping address. It will ship before the end of the week.

PolSerg, 2020-02-25:
> I have a new version 1.3 that I plan to sell on Tindie soon. It will probably be at least 4-5 weeks before they are ready due to the problems in China.
> 
> I have 3 hand built prototypes that I would be happy to sell for $60 each. They would not include drivers or ESP32.
> 
> [image]

Hello! And what is the difference from version 1.2, were there errors? I have already made a 1.2 board for myself, but have not yet soldered it, do I need to fix something to make it work?

bdring, 2020-02-25:
There is very little change between 1.2 and 1.3. 1.2 should work fine. These days there is very little additional cost to making changes between orders.  

I think the input voltage range is mis-labeled on the 1.2. Max voltage should be 24VDC. The limiting factor for the input voltage is the 5V power supply.

haklein, 2020-02-28:
@bdring do I need a special branch of Grbl_ESP32 to run this board with TMC5160 and SPI? Or has this been integrated in master?

haklein, 2020-02-28:
pls ignore my last comment, I found  CPU_MAP_SPI_DAISY_4X in the map on master

haklein, 2020-03-25:
I'm very impressed with your board. Also with your code. Thanks a lot! The steppers work fine on my 3 axes, but I'm unable to trigger the relay. The interlock jumper is set (connecting the both interlock pins). The JP5 close to the relay is also bridged.
One of the Relay pins seems to not be soldered. Is this intentional?

bdring, 2020-03-25:
All pins should be soldered.

Your settings are correct.

haklein, 2020-03-25:
I've soldered the pin but the relay is still not working. I've verified GPIO 25 (Spindle_3v3) with a multimeter, and it toggles from 0 to 3.3V when being turned on via M3. But that's not arriving as 5V signal at the interlock.

haklein, 2020-03-25:
Applying 5V to JP5 operates the relay, so apparently there is an issue with the TXS0101. From the datasheets, the voltages at the pins should be fine. But I'm unsure about the orientation of the package, mind to cross check?
[image]

bdring, 2020-03-25:
That orientation is correct. do you have both jumpers installed?

[image]

haklein, 2020-03-25:
Yes, I have both jumpers connected as shown in your picture. I just had removed the interlock when taking the picture as I was also measuring on those pins.

bdring, 2020-03-26:
Remove the jumpers and measure the voltage with spindle on at GPIO_NUM_25, GPIO_NUM_4 and the  inner side of the interlock jumper.

haklein, 2020-03-26:
With both jumpers removed, when spindle is on, I get the following readings:
G25: 3,18V
G4: 3,17V
Inner side (closer to C4 pad) Interlock: 0,002V

bdring, 2020-03-26:
Measure the the voltage on the J3 5V pin.

haklein, 2020-03-26:
J3 5V shows 4,95V

bdring, 2020-03-26:
It seems like the level shifter is not working.  If you are just using the spindle, you could try to bypass the level shifter. The relay will work on 3.3V.

haklein, 2020-03-26:
Ok, thanks for the hint, I will try that. Can you tell me the exact part number for the shifter (same package/footprint etc), then I'll attempt to replace it.

bdring, 2020-03-26:
Texas Instriments TXS0101

haklein, 2020-03-26:
the TXS0101DBV as SOT-23 (6)?

bdring, 2020-03-26:
Yes.

If you want to bypass it, just remove it and bridge these 2 pins.

[image]

haklein, 2020-03-27:
I've connected GPIO25 to the right pin of JP5 with a wire, relay works with G3/G5 now. That will do fine until the TXS0101DBV arrives. Thanks for the hint.

haklein, 2020-04-16:
The replacement TXS0101DBV did arrive today. After replacement, the relay did still not work. In the end it was the solder joints of R1, apparently one was cold. Re-heating made the relay work. Thanks for your support and the nice board.