bdring/4_Axis_SPI_CNC #7: BOM anywhere?

gitxmax, 2020-07-08:
I'm far from US, so ordering the pre-made board is next to impossible.
Uploaded the gerber to Jlcpcb and found they can do SMT.
The problem is I went through every corner of this repository but failed to find the components list.
Any hint?
Thank you!

bdring, 2020-07-08:
Here is a PDF. 

Note: JLCPCB has a very restrictive list of parts. I have never been able to use their service.

[SPI_4X V1.3 BOM.pdf](https://github.com/bdring/4_Axis_SPI_CNC/files/4892292/SPI_4X.V1.3.BOM.pdf)

gitxmax, 2020-07-10:
Thank you.
Went throgh the BOM, seems the following 2 components, might be a bit hard to get for me:

TXS0101DBVR - Level shifter
MMBT9013H - Transistor

Can this be replaced with something similar?