bdring/4_Axis_SPI_CNC #10: SD card not recognised

DrJohnM61, 2020-08-20:
I have tried plugging in an SD card (128mb) and tried a number of different formats (FAT12 and FAT), however, the ESP_32 web interface reports that there is no SD card. Is there a trick to getting the SD card to be recognised (I checked the code and the SD card option is tuned on). Does there are need to be in some other format (eg EXT)?

DrJohnM61, 2020-08-20:
By way of an update, the actual message in the web interface is "No SD Card". I also just tried the sd card formatted as ext4 without any joy.

bdring, 2020-08-20:
The wiki has some suggestions.

https://github.com/bdring/Grbl_Esp32/wiki/Using-the-SD-Card#card-formatting

DrJohnM61, 2020-08-20:
> The wiki has some suggestions.
> 
> https://github.com/bdring/Grbl_Esp32/wiki/Using-the-SD-Card#card-formatting

Thank for directing me to that page. I am running on a Mac and also Raspberry pi, so that formatting solution did not work. I swapped out the card for a different one and that is recognised. On the basis that the no-working card is 128mb (yes mb, not gb), I suspect that it may be that the ESP-32 just does not like the card that is probably quite old.

Anyway - seems like I have a solution... Thanks for your help.

closed 2020-08-20 by DrJohnM61