bdring/4_Axis_SPI_CNC #12: FluidNC config

haklein, 2021-11-29:
Is there a ready-to-use example config for this board in FluidNC?

jamhot1, 2022-03-26:
Me also. I am trying the standard 4axis config in the fluidnc example and can't get a motor to spin. I've used grbl for a while and this config file concept is confusing me a little.

boucz, 2022-03-26:
here is my config file, 3 axes (jumper shorting the fourth driver), tmc5160
`board: 4 Axis SPI Daisy
name: 4 Axis SPI Daisy

arc_tolerance_mm: 0.002

stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 500.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
      allow_single_axis: true

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.36:low
      hard_limits: false
      pulloff_mm:1.000
      tmc_5160:
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.075
        run_amps: 2.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 4
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 3
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: Stallguard
        use_enable: false
        step_pin: gpio.12
        direction_pin: gpio.14
        disable_pin: NO_PIN

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 900.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 600.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 600.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 1.100
      allow_single_axis: true

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.39:low
      hard_limits: false
      pulloff_mm:1.000
      tmc_5160:
        spi_index: 2
        r_sense_ohms: 0.075
        run_amps: 1.200
        hold_amps: 0.500
        microsteps: 16
        stallguard: 15
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 3
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: Stallguard
        use_enable: false
        step_pin: gpio.27
        direction_pin: gpio.26
        disable_pin: NO_PIN

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 700.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 300.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 100.000
      seek_scaler: 1.100
      feed_scaler: 1.100
      allow_single_axis: true

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.34:low
      hard_limits: false
      pulloff_mm:1.000
      tmc_5160:
        spi_index: 3
        r_sense_ohms: 0.075
        run_amps: 1.200
        hold_amps: 1.000
        microsteps: 16
        stallguard: 15
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 3
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: Stallguard
        use_enable: false
        step_pin: gpio.15
        direction_pin: gpio.2
        disable_pin: NO_PIN

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: NO_PIN

probe:
  pin: gpio.35:low
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

PWM:
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: gpio.4
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 12000=100%
`

jamhot1, 2022-03-26:
Thanks! Much appreciated! I'll be using the 2109 drivers in my build. Will this be compatible? Once I've got some motion I can start to experiment with the settings.
Thanks again

boucz, 2022-03-26:
Do you mean TMC2209? If so, you'll have a problem as on this board ESP32 uses SPI to speak to drivers (TMC2130, TMC5160 compatible). TMC2209 are serial drivers, you'll need a different board to use them.

jamhot1, 2022-03-27:
Yes. The tmc2209. Damn I cancelled my order for the 2130 in favour of these. I'll get some 5160's. At least it will be a clone of your current setup and I can start to figure the rest out. Thanks again for the pointers.
Ps, when you say serial. Is that described as UART. In the product description?